linux-nvme/nvme-cli #2791: nvme cli hangs when trying to set timestamp with set-features

tjodork, 2025-04-30:
If I enter nvme set-feature /dev/nvme7 -f 14 -v 1746041687000  and hit enter the command starts to execute but NEVER returns

ONCE I hit enter again ..it returns with 

set-feature:0x0e (Timestamp), value:00000000, cdw12:00000000, save:0
       0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
0000: 0a 00 00 00 00 00 00 00                          "........"

The level of nvme is   nvme-2.11

ikegami-t, 2025-05-01:
You need to set the timestamp data below as the data option argument.
[image]
Note: The data below just sent the LF (line break) code as hit the enter.
> 0000: 0a 00 00 00 00 00 00 00 "........"

ikegami-t, 2025-05-01:
By the way just created the PR: #2792 to add `nvme feat timestamp` command. Could you please test the command also if you have a chance? The command can be set the `tstmp` parameter for the timestamp value instead of the above timestamp data for the `nvme set-feature` command.

tjodork, 2025-05-01:
Thanks so much for giving me detail on this.. 
Clearly something changed right?  as this used to work.
I'm not good at recompiling nvme ...so it may be a bit before I try your new feature...but it will be very helpful

one other thing... why did the CLI hang vs returning an error ?

ikegami-t, 2025-05-02:
> Clearly something changed right? as this used to work.

No in my understanding the `nvme set-feature` behavior should be same from before. To makre sure can you test the older nvme-cli version? If needed I can check the implementation changes related to the issue.

> I'm not good at recompiling nvme ...so it may be a bit before I try your new feature...but it will be very helpful

I think you can use the AppImage built as below.
  <https://github.com/linux-nvme/nvme-cli/suites/37954605470/artifacts/3044303945>
    Note: You may need to install the following FUSE to execute the AppImage.
      <https://github.com/AppImage/AppImageKit/wiki/FUSE>

> one other thing... why did the CLI hang vs returning an error ?

What error returned? By the way you set the `-v` parameter and the parameter value is used for the command dword 11 but the timestamp feature does not use the dword value so looks the command invalid operation I think.
> nvme set-feature /dev/nvme7 -f 14 -v 1746041687000

tjodork, 2025-05-05:
i am positive that it did not hang before.

what error returned ?.  it hung... when doing below ..

    nvme set-feature /dev/nvme7 -f 14 -v 1746041687000

thanks to your help its all working.

ikegami-t, 2025-05-06:
> i am positive that it did not hang before.

Just confirmed the implementation as you are correct. Now the `-v` option was changed to `-V` as for the `--value' option alias. (Currenty the `-v` option is alias for the `verbose` option.) Very sorry for that changes. So please use `-V` or `--value' options instead.
```sh
nvme set-feature /dev/nvme7 -f 14 -v 1746041687000
```
  Note: This issue related to the issue: #2734.

Also the `--value' option (including `-V`) can be used for the set feature command timestamp feature (FID: 0Eh) setting.
```c
static int set_feature(int argc, char **argv, struct command *cmd, struct plugin *plugin)
{
...
	if (buf) {
		/*
		 * Use the '-v' value for the timestamp feature if provided as
		 * a convenience since it can often fit in 4-bytes. The user
		 * should use the buffer method if the value exceeds this
		 * length.
		 */
		if (cfg.feature_id == NVME_FEAT_FID_TIMESTAMP &&
		    argconfig_parse_seen(opts, "value")) {
			memcpy(buf, &cfg.value, NVME_FEAT_TIMESTAMP_DATA_SIZE);
```

igaw, 2025-05-07:
Yes, we decided late in the 2.x cycle to introduce a couple of common command line flags. This was a mistake as it broke a lot of workflows, but changing back was also not really an option. So we stuck with it. Sorry about that.

When nvme-cli doesn't get a value via the command line '-V' it wants to read it from stdin. That's why it hangs.

tjodork, 2025-05-07:
thanks for the clarification and glad that I can use the --value option ...the data buffer method is not ideal   : )

igaw, 2025-05-07:
FWIW, we are now very carefully not to break more stuff. Sure there will be more going wrong but hopefully not so big as the `-v` and `-o` global option. From what I understand, it works with `--value`? If so, good to close?

tjodork, 2025-05-07:
yes !

closed 2025-05-07 by igaw