linux-nvme/nvme-cli #2800: nvme connect-all exits with status 0 despite failed discovery or connection attempts

arun-kv, 2025-05-08:
[image]

nvme -version
nvme version 2.3 (git 2.3)
libnvme version 1.3 (git 1.3)

This was not the case for older versions. Because of this nvmf-autoconnect.service is not working as expected.

igaw, 2025-05-08:
With version was it working? 

The connect-all code will ignore errors for individual connects and keeps continuing to connect to the remaining connections, it might issue a warning for 'already connected', the rest will be eaten up.

nvmf-autoconnect is a one shot attempt to connect to all available services, systemd is not aware of any individual connection are their state. This is really a crutch, the proper approach is to use a unit file per connection.

arun-kv, 2025-05-08:
> With version was it working?
>
> The connect-all code will ignore errors for individual connects and keeps continuing to connect to the remaining connections, it might issue a warning for 'already [...]

For nvme version 1.16, it was working fine.

I had multiple target entries in the /etc/nvme/discovery.conf, even when connect fails for one target the exist status for connect-all was non zero. and i have modified the nvmf-autoconnect service with Restart=on-failure. This way service retry the connect-all untill all the connections are successful.

igaw, 2025-05-08:
The 1.x branch iterates over all entries in discovery.conf and if one fails it keeps track of the error and returns that one.
In the 2.x branch all return code from `__discover` are ignored.

igaw, 2026-09-24:
Fixes for nvme-cli 2.x is merged.

closed 2026-09-24 by igaw