linux-nvme/nvme-cli #2811: nvme-cli improvement for the lbaf related info with json output and --verbose

yizhanglinux, 2025-05-13:
It's better to output the actual data size with json format and verbose params.
```
# nvme id-ns --verbose  /dev/nvme0n1
---snip---
LBA Format  0 : Metadata Size: 0   bytes - Data Size: 512 bytes - Relative Performance: 0x1 Better
LBA Format  1 : Metadata Size: 8   bytes - Data Size: 512 bytes - Relative Performance: 0x3 Degraded (in use)
LBA Format  2 : Metadata Size: 0   bytes - Data Size: 4096 bytes - Relative Performance: 0 Best
LBA Format  3 : Metadata Size: 8   bytes - Data Size: 4096 bytes - Relative Performance: 0x2 Good
LBA Format  4 : Metadata Size: 64  bytes - Data Size: 4096 bytes - Relative Performance: 0x3 Degraded
```

```
# nvme id-ns --verbose --output-format=json   /dev/nvme0n1 | jq ".lbafs"
[
  {
    "ms": 0,
    "ds": 9,
    "rp": 1
  },
  {
    "ms": 8,
    "ds": 9,
    "rp": 3
  },
  {
    "ms": 0,
    "ds": 12,
    "rp": 0
  },
  {
    "ms": 8,
    "ds": 12,
    "rp": 2
  },
  {
    "ms": 64,
    "ds": 12,
    "rp": 3
  }
]
```

ikegami-t, 2025-05-17:
Just created the PR: #2817 for this enhancement and it changes the outputs as below then is it okay for you?
  Note: This changes to add both `lbaf` and `in_use` parameters output for both verbose and nomal modes.
```sh
tokunori@tokunori-desktop:~/nvme-cli$ nvme-build id-ns /dev/nvme0n1 -o json | jq ".lbafs"
[
  {
    "lbaf": 0,
    "ms": 0,
    "ds": 9,
    "rp": 0,
    "in_use": 1
  }
]
tokunori@tokunori-desktop:~/nvme-cli$ nvme-build id-ns /dev/nvme0n1 -o json -v | jq ".lbafs"
[
  {
    "LBA Format": 0,
    "Metadata Size": "0 bytes",
    "Data Size": "512 bytes",
    "Relative Performance": "0x0 Best",
    "in use": "yes"
  }
]
```

ikegami-t, 2025-05-20:
Just released the v2.14 applied the PR: #2817 changes. Thank you.

yizhanglinux, 2025-05-20:
Thanks for the improvement. Close this issue now.