linux-nvme/nvme-cli #2814: nvme copy command show same output for all the output format

bhaviks-company, 2025-05-15:
nvme lib and version

```
libnvme-1.9-3.el9.x86_64
nvme-cli-2.9.1-6.el9.x86_64

nvme -version
nvme version 2.9.1 (git 2.9.1)
libnvme version 1.9 (git 1.9)
```

While trying to run nvme copy command with different output format, there is no change in output format for success message.

```

nvme copy -v -n 6 -d 0x3 -b 1 -F 0 /dev/nvme3n6 -o json
latency      : 11851 us
NVMe Copy: success

nvme copy -v -n 6 -d 0x3 -b 1 -F 0 /dev/nvme3n6 -o normal
latency      : 18678 us
NVMe Copy: success

nvme copy -v -n 6 -d 0x3 -b 1 -F 0 /dev/nvme3n6 -o binary
latency      : 17795 us
NVMe Copy: success
```

Although error message is correctly formatted.
```
# nvme copy -v -n 4 -d 0x3 -b 1 -F 0 /dev/nvme3n6 -o json
latency      : 10278 us
{
  "error":"NVMe Copy: Invalid argument"
}
```

igaw, 2025-05-16:
IIRC, this should be fixed with the current release. Could you try with the latest version and see if this is still the case?

ikegami-t, 2025-05-18:
Just confirmed the implementation so looks same behavior for the current release. Also the latency output is print to stdout only.
1. Copy command
```c
static int copy_cmd(int argc, char **argv, struct command *cmd, struct plugin *plugin)
{
...
	err = nvme_copy(&args);
	if (err < 0)
		nvme_show_error("NVMe Copy: %s", nvme_strerror(errno));
	else if (err != 0)
		nvme_show_status(err);
	else
		printf("NVMe Copy: success\n");
```
2. Latency ouput
```c
static void nvme_show_latency(struct timeval start, struct timeval end)
{
	printf("latency      : %lu us\n",
	       (end.tv_sec - start.tv_sec) * 1000000 + (end.tv_usec - start.tv_usec));
```
Note: The latency output changed on the current release as output by `-vv` but not `-v`.

ikegami-t, 2025-05-18:
Just created the PR: #2818 then the output will be changed as below is this okay you?
```sh
# nvme copy -v -n 4 -d 0x3 -b 1 -F 0 /dev/nvme3n6 -o json
latency      : 10278 us
{
  "NVMe Copy":"success"
}
```
Note: Still I am changing for the latency output also for the json format.

ikegami-t, 2025-05-18:
Added commits in the PR for the latency output then the output will be like as below but I am not sure if okay to use the curly brackets separately used for the command result and the latency (and the command debug) then two json objects output so if needed I will change the json format again for the lateny output.
```sh
tokunori@tokunori-desktop:~/nvme-cli$ nvme-build id-ns /dev/nvme0n1 -vv -o json
{
  "nsze":2000409264,
  "ncap":2000409264,
  "nuse":2000409264,
...
  "lbafs":[
    {
      "ms":0,
      "ds":9,
      "rp":0
    }
  ],
  "vs":[]
}
{
  "opcode":"06",
  "flags":"00",
  "rsvd1":"0000",
...
  "cdw14":"00000000",
  "cdw15":"00000000",
  "timeout_ms":"00000000",
  "result":"00000000",
  "err":"0",
  "latency":"5100 us"
}
```
Note: The examble above for the id-ns command but not for the copy command but the `-vv` outouts for latency and command debug (opcode, etc.) will be same.

ikegami-t, 2025-05-20:
Just released the v2.14 applied the PR: #2818 changese. Thank you.