linux-nvme/nvme-cli #2834: Nvme list displays wrong size when size is large

akshimmu, 2025-06-02:
I am working with a device that supports upto 256TB.
When i run nvme list, i see that the size reported is wrong.

It shows 281.47TB instead of 256TB.

NSZE is 0x1000000000. LBA Size-4096.

This is wrong because the size being advertised is larger than the size available.

[image]

igaw, 2025-06-02:
Not sure what you want to say with the image.

The list output is generated here:

```c
static void stdout_list_item(nvme_ns_t n)
{
	char usage[128] = { 0 }, format[128] = { 0 };
	char devname[128] = { 0 }; char genname[128] = { 0 };

	long long lba = nvme_ns_get_lba_size(n);
	double nsze = nvme_ns_get_lba_count(n) * lba;
	double nuse = nvme_ns_get_lba_util(n) * lba;

	const char *s_suffix = suffix_si_get(&nsze);
	const char *u_suffix = suffix_si_get(&nuse);
	const char *l_suffix = suffix_binary_get(&lba);

	snprintf(usage, sizeof(usage), "%6.2f %2sB / %6.2f %2sB", nuse,
		u_suffix, nsze, s_suffix);
```

If I read it correctly, it returns the size in bytes. And git history says, this how it is done since the beginning.

akshimmu, 2025-06-03:
[image]

Doesnt this mean that the device supports 281.47TB(i.e Lbas upto 281.47TB) ?
When in reality, lbas are supported only upto 256TB?

akshimmu, 2025-06-04:
> This comes down to how `TB` is understood in this context. I suppose we should at least document this in the man pages for nvme list.

I think this would be good

ikegami-t, 2025-06-04:
Just pushed the PR #2837 to document this in the man pages.

igaw, 2025-06-30:
PR merged. Thanks you @ikegami-t