linux-nvme/nvme-cli #2867: Total nvme capacity seems wrong

akshimmu, 2025-07-16:
Total capacity should be 1GB. But it displays 8 GB.

[image]

akshimmu, 2025-07-16:
[image]

[image]

igaw, 2025-07-16:
Which version of nvme-cli/libnvme are you using? There was a bug which resulted showing the wrong size, but that has been fixed a while ago. If you are using an older version, please retest with latest released version.

igaw, 2025-07-17:
https://github.com/linux-nvme/nvme-cli/wiki/FAQ#nvme-list-shows-wrong-diskblock-size

closed 2025-07-17 by igaw