linux-nvme/nvme-cli #2868: wdc/sndk plug-in not functioning as expected for SanDisk SN861.

Sayoojntly, 2025-07-16:
Steps

1. Power on the NVME OF enclosure with SN861 drives connected to host.
2. Check the wdc/sndk plug in commands.

Observations:
WDC Plug-in not functioning as expected
```
nvme wdc get-drive-status /dev/nvme1
ERROR: WDC: wdc_get_pci_ids : Open vendor file failed
ERROR: WDC: wdc_get_pci_ids : Open vendor file failed
ERROR: WDC: wdc_get_dev_mng_log_entry: Detected unaligned end of the data. Data Offset: 0x8 Entry Size: 0x0, Remaining Log Length: 0xba Entry Id: 0x0
ERROR: WDC: C2 log id 0x8 not found with uuid index 0
Not found, requesting log page with uuid_index 0
ERROR: WDC: wdc_get_dev_mng_log_entry: Detected unaligned end of the data. Data Offset: 0x8 Entry Size: 0x0, Remaining Log Length: 0xba Entry Id: 0x0
ERROR: WDC: C2 log id 0x8 not found with uuid index 0
ERROR: WDC: 0xC2 Log Page entry ID 0x8 not found
ERROR: WDC: wdc_get_pci_ids : Open vendor file failed
ERROR: WDC: wdc_get_dev_mng_log_entry: Detected unaligned end of the data. Data Offset: 0x8 Entry Size: 0x0, Remaining Log Length: 0xba Entry Id: 0x0
ERROR: WDC: C2 log id 0x8 not found with uuid index 0
Not found, requesting log page with uuid_index 0
ERROR: WDC: wdc_get_dev_mng_log_entry: Detected unaligned end of the data. Data Offset: 0x8 Entry Size: 0x0, Remaining Log Length: 0xba Entry Id: 0x0
ERROR: WDC: C2 log id 0x8 not found with uuid index 0
ERROR: WDC: 0xC2 Log Page entry ID 0x8 not found
ERROR: WDC: wdc_get_pci_ids : Open vendor file failed
ERROR: WDC: wdc_get_dev_mng_log_entry: Detected unaligned end of the data. Data Offset: 0x8 Entry Size: 0x0, Remaining Log Length: 0xba Entry Id: 0x0
ERROR: WDC: C2 log id 0x8 not found with uuid index 0
Not found, requesting log page with uuid_index 0
ERROR: WDC: wdc_get_dev_mng_log_entry: Detected unaligned end of the data. Data Offset: 0x8 Entry Size: 0x0, Remaining Log Length: 0xba Entry Id: 0x0
ERROR: WDC: C2 log id 0x8 not found with uuid index 0
ERROR: WDC: 0xC2 Log Page entry ID 0x8 not found
ERROR: WDC: wdc_get_pci_ids : Open vendor file failed
ERROR: WDC: wdc_get_dev_mng_log_entry: Detected unaligned end of the data. Data Offset: 0x8 Entry Size: 0x0, Remaining Log Length: 0xba Entry Id: 0x0
ERROR: WDC: C2 log id 0x8 not found with uuid index 0
Not found, requesting log page with uuid_index 0
ERROR: WDC: wdc_get_dev_mng_log_entry: Detected unaligned end of the data. Data Offset: 0x8 Entry Size: 0x0, Remaining Log Length: 0xba Entry Id: 0x0
ERROR: WDC: C2 log id 0x8 not found with uuid index 0
ERROR: WDC: 0xC2 Log Page entry ID 0x8 not found
ERROR: WDC: wdc_get_pci_ids : Open vendor file failed
ERROR: WDC: wdc_get_dev_mng_log_entry: Detected unaligned end of the data. Data Offset: 0x8 Entry Size: 0x0, Remaining Log Length: 0xba Entry Id: 0x0
ERROR: WDC: C2 log id 0x15 not found with uuid index 0
Not found, requesting log page with uuid_index 0
ERROR: WDC: wdc_get_dev_mng_log_entry: Detected unaligned end of the data. Data Offset: 0x8 Entry Size: 0x0, Remaining Log Length: 0xba Entry Id: 0x0
ERROR: WDC: C2 log id 0x15 not found with uuid index 0
wdc_get_fw_cust_id: ERROR: WDC: 0xC2 Log Page entry ID 0x15 not found
wdc_get_enc_drive_capabilities: ERROR: WDC: invalid customer id
ERROR: WDC: wdc_get_pci_ids : Open vendor file failed
ERROR: WDC: wdc_get_dev_mng_log_entry: Detected unaligned end of the data. Data Offset: 0x8 Entry Size: 0x0, Remaining Log Length: 0xba Entry Id: 0x0
ERROR: WDC: C2 log id 0x8 not found with uuid index 0
Not found, requesting log page with uuid_index 0
ERROR: WDC: wdc_get_dev_mng_log_entry: Detected unaligned end of the data. Data Offset: 0x8 Entry Size: 0x0, Remaining Log Length: 0xba Entry Id: 0x0
ERROR: WDC: C2 log id 0x8 not found with uuid index 0
ERROR: WDC: 0xC2 Log Page entry ID 0x8 not found
ERROR: WDC: 0xC2 Log Page not supported

```

jeff-lien-sndk, 2025-07-16:
I see 2 issues here.

1. The "wdc_get_pci_ids : Open vendor file failed" messages are caused by the drive being fabric attached.  We're unable to get the the pci device id's in fabric attached drives.  This issue can be ignored though.  It generates noise with the several additional error messages per command but they don't appear to be causing any functional issues.
2. Using uuid_index of 0 to read the WDC C2 log page is not correct.  I'm not sure why an index of 0 is being used.  The code wll search the uuid list for the WDC uuid and will determine the correct index to use based on that as per how it's stated in the nvme spec.  This is working as expected for me on a direct attached SN861.  

@Sayoojntly I will continue to look into the wdc and sandisk plugin code but have a few questions for you.  

- What's the pci device id of the SN861 drive you're using?  
- What's the version of FW loaded on it?  
- Could you execute the command "nvme id-uuid /dev/nvmeX" on your drive and post the output here?

@igaw  Are you aware of an nvme cli command limitations or restrictions with nvme over fabric attached drives?

Sayoojntly, 2025-07-17:
Hi @jeff-lien-sndk ,
Please find my response.

- What's the pci device id of the SN861 drive you're using?
@Sayoojntly The drive is connected through NVMe OF enclosure. 
```
Device           Cntlid SN                   MN                                       FR       TxPort Address        Slot   Subsystem    Namespaces
---------------- ------ -------------------- ---------------------------------------- -------- ------ -------------- ------ ------------ ----------------
nvme0    16386  240562800020         WUS5EA138ESP7E1                          RC610008 rdma   traddr=20.20.20.6,trsvcid=4420        nvme-subsys0
nvme1    16396  37241CE0000310       SDS6BA138PSP9X1                          RH110013 rdma   traddr=10.10.10.6,trsvcid=4420        nvme-subsys1 nvme1n1
nvme10   16388  372415E0006610       SDS6BA138PSP9X1                          RH110013 rdma   traddr=10.10.10.6,trsvcid=4420        nvme-subsys10 nvme10n1
```

- What's the version of FW loaded on it?
@Sayoojntly The drive firmware is **RH110013**
```
nvme list
Node                  Generic               SN                   Model                                    Namespace  Usage                      Format           FW Rev
--------------------- --------------------- -------------------- ---------------------------------------- ---------- -------------------------- ---------------- --------
/dev/nvme10n1         /dev/ng10n1           372415E0006610       SDS6BA138PSP9X1                          0x1          1.27  TB /   3.83  TB      4 KiB +  0 B   RH110013
/dev/nvme11n1         /dev/ng11n1           37241CE0000210       SDS6BA138PSP9X1                          0x1          1.25  TB /   3.83  TB      4 KiB +  0 B   RH110013
```

- Could you execute the command "nvme id-uuid /dev/nvmeX" on your drive and post the output here?
@Sayoojntly The id-uuid is giving **Invalid Field in Command**
```
 nvme id-uuid /dev/nvme10
NVMe status: Invalid Field in Command: A reserved coded value or an unsupported value in a defined field(0x4002)
```

I am attaching id-ctrl from the drive
```
 nvme id-ctrl /dev/nvme10 -H
NVME Identify Controller:
vid       : 0x1b96
ssvid     : 0x1b96
sn        : 372415E0006610
mn        : SDS6BA138PSP9X1
fr        : RH110013
rab       : 2
ieee      : 0014ee
cmic      : 0x2
  [3:3] : 0     ANA not supported
  [2:2] : 0     PCI
  [1:1] : 0x1   Multi Controller
  [0:0] : 0     Single Port

mdts      : 8
cntlid    : 0x4004
ver       : 0x10400
rtd3r     : 0x1c9c380
rtd3e     : 0x4c4b40
oaes      : 0x300
  [31:31] : 0   Discovery Log Change Notice Not Supported
  [27:27] : 0   Zone Descriptor Changed Notices Not Supported
  [16:16] : 0   Temperature Threshold Hysteresis Recovery Not Supported
  [15:15] : 0   Normal NSS Shutdown Event Not Supported
  [14:14] : 0   Endurance Group Event Aggregate Log Page Change Notice Not Supported
  [13:13] : 0   LBA Status Information Notices Not Supported
  [12:12] : 0   Predictable Latency Event Aggregate Log Change Notices Not Supported
  [11:11] : 0   Asymmetric Namespace Access Change Notices Not Supported
  [9:9] : 0x1   Firmware Activation Notices Supported
  [8:8] : 0x1   Namespace Attribute Changed Event Supported

ctratt    : 0x40
  [19:19] : 0   Flexible Data Placement Not Supported
  [17:17] : 0   HMB Restrict Non-Operational Power State Access Not Supported
  [16:16] : 0   MDTS and Size Limits Exclude Metadata Not Supported
  [15:15] : 0   Extended LBA Formats Not Supported
  [14:14] : 0   Delete NVM Set Not Supported
[... 288 lines cut ...]
ERROR: WDC: wdc_get_dev_mng_log_entry: Detected unaligned end of the data. Data Offset: 0x8 Entry Size: 0x0, Remaining Log Length: 0xba Entry Id: 0x0
ERROR: WDC: C2 log id 0x8 not found with uuid index 0
Not found, requesting log page with uuid_index 0
ERROR: WDC: wdc_get_dev_mng_log_entry: Detected unaligned end of the data. Data Offset: 0x8 Entry Size: 0x0, Remaining Log Length: 0xba Entry Id: 0x0
ERROR: WDC: C2 log id 0x8 not found with uuid index 0
ERROR: WDC: 0xC2 Log Page entry ID 0x8 not found
ERROR: WDC: wdc_get_pci_ids : Open vendor file failed
ERROR: WDC: wdc_get_dev_mng_log_entry: Detected unaligned end of the data. Data Offset: 0x8 Entry Size: 0x0, Remaining Log Length: 0xba Entry Id: 0x0
ERROR: WDC: C2 log id 0x15 not found with uuid index 0
Not found, requesting log page with uuid_index 0
ERROR: WDC: wdc_get_dev_mng_log_entry: Detected unaligned end of the data. Data Offset: 0x8 Entry Size: 0x0, Remaining Log Length: 0xba Entry Id: 0x0
ERROR: WDC: C2 log id 0x15 not found with uuid index 0
wdc_get_fw_cust_id: ERROR: WDC: 0xC2 Log Page entry ID 0x15 not found
wdc_get_enc_drive_capabilities: ERROR: WDC: invalid customer id
ERROR: WDC: wdc_get_pci_ids : Open vendor file failed
ERROR: WDC: wdc_get_dev_mng_log_entry: Detected unaligned end of the data. Data Offset: 0x8 Entry Size: 0x0, Remaining Log Length: 0xba Entry Id: 0x0
ERROR: WDC: C2 log id 0x8 not found with uuid index 0
Not found, requesting log page with uuid_index 0
ERROR: WDC: wdc_get_dev_mng_log_entry: Detected unaligned end of the data. Data Offset: 0x8 Entry Size: 0x0, Remaining Log Length: 0xba Entry Id: 0x0
ERROR: WDC: C2 log id 0x8 not found with uuid index 0
ERROR: WDC: 0xC2 Log Page entry ID 0x8 not found
ERROR: WDC: 0xC2 Log Page not supported
```

@jeff-lien-sndk Please let me know anything more required from my side.

jeff-lien-sndk, 2025-07-17:
@Sayoojntly I see uuid lists are not supported on that version of SN861.  I'll need to check into this more to determine if there's a way to fix it.

jeff-lien-sndk, 2025-07-24:
I've asked our FW team about this and am still waiting on response from them.

jeff-lien-sndk, 2025-07-28:
The FW team responded with "The SN861 product manual states that it does not support NVMe over fabrics".  If this is a new requirement, you need to work with the Customer Support Team to get it raised to the FW team.

sarabjotsoit, 2025-07-31:
Hi, @jeff-lien-sndk 

We have checked with our SanDisk POC, and according to him, the Support is the same as that of the Ultrastar DC SN650 and SN655 NVMe SSDs.

Is there anything specific we are looking for ?

closed 2025-10-07 by igaw