linux-nvme/nvme-cli #2876: Replace AppImage artifcat with the static build artifact

igaw, 2025-07-24:
The AppImage build started to fail because of broken dependencies. Let's replace this with the static build artifact which is way simpler to handle from our side.

ikegami-t, 2025-07-24:
Is the actual build failure mentioned below for the PR #2875 appimage check failed?
  <https://github.com/linux-nvme/nvme-cli/actions/runs/16491703761/job/46627720769?pr=2875>
```sh
INFO:root:Downloading: https://github.com/AppImage/AppImageKit/releases/download/continuous/runtime-x86_64
Traceback (most recent call last):
  File "/usr/local/bin/appimage-builder", line 8, in <module>
    sys.exit(__main__())
  File "/usr/local/lib/python3.8/dist-packages/appimagebuilder/__main__.py", line 50, in __main__
    invoker.execute(commands)
  File "/usr/local/lib/python3.8/dist-packages/appimagebuilder/invoker.py", line 29, in execute
    command()
  File "/usr/local/lib/python3.8/dist-packages/appimagebuilder/commands/create_appimage.py", line 27, in __call__
    self.primer.prime()
  File "/usr/local/lib/python3.8/dist-packages/appimagebuilder/modules/prime/appimage_primer.py", line 43, in prime
    self._get_appimage_kit_runtime()
  File "/usr/local/lib/python3.8/dist-packages/appimagebuilder/modules/prime/appimage_primer.py", line 101, in _get_appimage_kit_runtime
    request.urlretrieve(url, self.carrier_path)
  File "/usr/lib/python3.8/urllib/request.py", line 247, in urlretrieve
    with contextlib.closing(urlopen(url, data)) as fp:
  File "/usr/lib/python3.8/urllib/request.py", line 222, in urlopen
    return opener.open(url, data, timeout)
  File "/usr/lib/python3.8/urllib/request.py", line 531, in open
    response = meth(req, response)
  File "/usr/lib/python3.8/urllib/request.py", line 640, in http_response
    response = self.parent.error(
  File "/usr/lib/python3.8/urllib/request.py", line 569, in error
    return self._call_chain(*args)
  File "/usr/lib/python3.8/urllib/request.py", line 502, in _call_chain
    result = func(*args)
  File "/usr/lib/python3.8/urllib/request.py", line 649, in http_error_default
    raise HTTPError(req.full_url, code, msg, hdrs, fp)
urllib.error.HTTPError: HTTP Error 404: Not Found
```
It seems the failure caused by the download failure and also actually the link below failed to open as page not found so looks not related the build failure to the dynamic build.
  <https://github.com/AppImage/AppImageKit/releases/download/continuous/runtime-x86_64>

igaw, 2025-07-24:
All builds fail because the container has been removed. I tried to replace it with a newer image but found out that all the AppImage actions are highly complex with a lot of dependencies. Last time I spend way to much time to setup this workflow. The reason why we have the AppImage workflow is to provide a standalone binary for testing. Now that the static builds works fine, we are able to drop AppImage and just provide the static build binary. This simplifies the setup considerable and avoids these huge dependency changes.

ikegami-t, 2025-07-24:
Unsderstood. Thank you.

igaw, 2025-07-28:
https://github.com/AppImage/AppImageKit/issues/1395

closed 2025-07-29 by igaw