linux-nvme/nvme-cli #2886: Missing "Controllers" section in `nvme list -o json -v`

tbzatek, 2025-07-31:
Caused by commit 64bed0a (Cc: @shroffni)

before: [before.txt](https://github.com/user-attachments/files/21533161/before.txt)
after: [after.txt](https://github.com/user-attachments/files/21533164/after.txt)

(still discovering regressions after building the latest releases in Fedora and retesting our stuff...)

We've been parsing `nvme list -o json -v` to extract various fields in our scripts. Any change to the JSON output structure is unfortunate but I understand it's needed to make things right. (I'm not even sure if this is supposed to be covered by any output stability guarantees).

However missing the whole _"Controllers"_ section is weird.

igaw, 2025-07-31:
Darn, I was hoping this one will not cause troubles. Anyway, we have `--output-format-version` for this. That means the default version v1 is needs to be behave before this change and v2 is allowed to do the new thing.

igaw, 2025-08-01:
Interesting idea, I am fine with at least try it out and see how many people are starting to complain :)

igaw, 2025-08-06:
Should be fixed with https://github.com/linux-nvme/nvme-cli/commit/80ad62ba75f1f1dd5b99c6f56769141b48344485